Alexbox364/F_Interface_AL #1: Can you help me? There is something unclear about the schematic diagram

jz5588, 2023-09-27:
[image]

What is the connection between the marked position of the picture? What is the function of flying wires? I can’t understand the schematic diagram here. It would be perfect if the schematic diagram has the connected devices marked.

Alexbox364, 2023-09-28:
Hi!
Sure it’s unclear in the breadboard view.
There the ground pins row that are used for button group (bottom left of the red circle), the power supply pin (not really needed but there was som space left) and the ICSP connector of the arduino.
I’ll update the sketch to be more comprehensive.
Inbetween, you can click on a component on The breadboard view, then go to the PcB view and you’ll see where the component actually is.

jz5588, 2023-09-28:
Thank you for your help. Can this PCB be directly connected to FANATEC's wheel and connected to ARDUINO NANO's USB MICO interface for other brands of wheels?

Alexbox364, 2023-09-30:
Hi, 
for the Fanatec bases yes, you’ll just need a podium hub without its original PCB to get the QR and 12 pins connector.
For other wheels you can use the micro usb of arduino nano to connect to Pc but you’ll need additional software on the Pc to turn the serial output of arduino into joystick button presses.
Like sim hub, but I never tested it. It’s highly likely you’ll have to further develop the software on the arduino nano.
It’s something I have in mind to develop connection with sim hub but it’s not done yet.

jz5588, 2023-10-03:
I need 16 WS2812RGB as speed and semaphore prompt lights. Can it be realized through the 3P socket of DO-VCC-GND?

Alexbox364, 2023-10-07:
Hi, yes, you can use as much LEDs on the NPX port as you want (keeping in mind the required mount of current).
But you’ll be limited by the data sent by fanalab

closed 2023-11-22 by Alexbox364