Alexbox364/F_Interface_AL #2: I have another question for you, is it possible to directly use these two control buttons?

jz5588, 2023-10-07:
I have another question for you, is it possible to directly use these two control buttons?
[image]

jz5588, 2023-10-07:
Because according to your tutorial, you choose the modular button

Alexbox364, 2023-10-07:
Hi, sure, you can use them directly, they are managed independently from the 4 button groups. You’ll just need to solder the relevant resistors.
You can also use the DPAD pins as classic buttons.

jz5588, 2023-10-07:
There are already resistors on the PCB you designed. Do I need to add another resistor if I use this button directly?

Alexbox364, 2023-10-09:
Yes, just solder the resistors identified on the PCB. It’s just that if you don’t need specific group of functions on the PcB, you don’t need to solder all the components

closed 2023-11-22 by Alexbox364