Alexbox364/F_Interface_AL #3: Can you help me,I don't know how to remove the arduino bootloader.

Lele1494, 2023-11-05:
Thank you for your great work！
My arduino nano already have bootloader,how to remove it? 
If i burn firmware directly through the ICSP，bootloader will be overwritten？

I'm sorry, my English is very poor。I use translation software to write。
Looking forward to your reply。

Alexbox364, 2023-11-08:
Hi! 
to flash the firmware without the Arduino bootloader, you can use the link in the readme : https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP

Alexbox364, 2023-11-08:
The process is revertable : and is described in the « program the bootloader » paragraph
https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP#program-the-bootloader

Lele1494, 2023-11-12:
> 该过程是可还原的：并在“程序引导加载程序”第 https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP#program-the-bootloader 段中进行了描述

Thank you! Already done! It worked.
 Do you know which model of wheel can use the oled display？

Alexbox364, 2023-11-12:
Hi, great 👍🏻.
In fact, whatever the wheel ID you choose (among the ones using SPI to communicate with the base), you will have access to the same amount of buttons / LEDs / 7 segments.
If you hook up an Oled display, you can just display simple data sent by the base (the one the is displayed on the 3 x 7segments digits and the number of LEDs of the rev limiter), they can be configured in Fanalab.
These 3 digits are parsed by the F_interface code and you can display it on an OLED display.