Alexbox364/F_Interface_AL #5: Small question about the boost circuit

thomaswang04, 2024-01-11:
I was looking at the cirucit for the stepup voltage converter and was wondering why you didnt step up the wheels 4.5v pin instead of boosting the 3.3v one? I am currently editing the pcb to work in my current wheel design and I wanted to power a raspberry pi off of it to run a sim hub display and wanted to ask if you think that the boost converter could supply the everything plus the raspberry pi and screen which i have mesured to draw about 0.5 to 0.8 amps at any given moment.

Alexbox364, 2024-01-11:
Hi!
We did so in the project because the 3,3V is powered at first by the base. 
5V pin is powered later and too late to establish a SPi communication with the base.
If you wish to power your RPi and the screen, I would advise you to use the 5V pin and the USB 5V pin.
I don’t know how much currrent the base can provide safely but I wouldn’t expect too much of it.
I would suggest you to use a microcontroller to unlock the base FFB and handle the SPI communication and use the RPI to handle buttons and screen.

Alexbox364, 2024-01-11:
I don’t think the RPi alone can fire up quickly enough at base startup to unlock FFB.

thomaswang04, 2024-01-11:
I have been trying to use the 5v usb pin with the usb ground but it ends up having a weird interaction. Sometimes when the wheel is attatched the wheel goes into safe mode then reboots and does the spinning calibration thing. However sometimes I can get it working by plugging the Rpi in after the wheel connects. I think ill try boosting the 4.5v circuit for just the RPi and the 3.3 for the arduino. Thanks for your input!

closed 2024-02-04 by Alexbox364