Alexbox364/F_Interface_AL #7: questions about the 0.06 hardware and software

thomaswang04, 2024-03-02:
Hi alex,
Im back asking more questions again. Since I last asked a question a couple months ago I have been trying to design a board using your design to fit in my wheel, however this new one fits now.  

Im making a mercedes f1 wheel but im desinging it so it has 6 rotary encoders and 2 rotary swtiches. Would it be possible to use the output pins used for the 2 screens and two of the rotary swithces for the encoders? I somewhat understand the code so i should be able to duplicate the encoder code and add left and right button bits right?

Also your button box has 10 leds, is it possible to have 21?

Sorry for asking so many questions.

Alexbox364, 2024-03-02:
Hi!
Yes, for sure you can use the LCD pins or any other pin you don’t use to hook up your extra encoders. 
As encoders are already managed without interrupts in the code, you can just duplicate existing portion of code to add your extra encoders :-).
For the button box, you can use as many WS2812b LEDs as you want. I did put 8 on the PCB because it fits but I did myself preferably used a piece of 120 LEDs / m strip sticker on the PCB with 10 LEDs. 21 will also be fine :-).

thomaswang04, 2024-03-05:
Cool! I'll get that sorted when I get the chance. Does that mean you can control each WS2812b individually from the fanatec software?

Alexbox364, 2024-03-08:
Hi, no you cannot control the aRGB LEDs from Fanalab. 
What you get from information on the arduino is the individual status (on / off) of each of the 9 original supported LEDs of old Fanatec wheels.
With this information, you can light up as many aRGB LEDs hooked up to the Arduino.

with Fanalab, you can decide when to light up those 9 LEDS