Alexbox364/F_Interface_AL #8: street wheel model

Lele1494, 2024-03-06:
Thank you for your great work
Sorry,My English is not good, I am using translation software to write。
I got a question,I choose the type is NANO_WHIT_HEADER.When I defined the wheel model as P-HUB,The keypad don't work,Defined as 918, kepad work's very well.How does this need to be resolved？
 BTW,Have you made this  ATTINY_NO_HEADER  ？
It's smaller，I am very interested in this project。
Thank you again for your help

Alexbox364, 2024-03-08:
Hi, no problem 👍🏻.
Hum, that’s strange as they keypad presses are buttons presses like others. You did not change anything else in the code?
To debug it, you can observe what’s going on on the Miso SPI data by uncommenting the print_miso() function. You’ll know if the wheel actually sends d_pad data to the base.

Lele1494, 2024-03-10:
Thank you,I'll check it.
Even so, the 918 mode works perfectly。
I designed this, put the chips together directly, and it's being customized. I hope it works.
[image]

Lele1494, 2024-03-10:
Thank you,I'll check it.
Even so, the 918 mode works perfectly。
I designed this, put the chips together directly, and it's being customized. I hope it works.
[image]
The factory is making it, I will test whether it works.

Lele1494, 2024-03-18:
I'm back,it works!!
3D print shell,looks very well!
Without your code, I wouldn't have been able to do it. Thank you very much!

Alexbox364, 2024-03-19:
Hi mate, this looks very cool! 
you are using an ATMEGA329P chip?
This is very tiny and can be useful for many people. Please consider sharing it on GitHub :-).
I’m working on Bluetooth based DIY and for Fanatec owners, that could be great to use you design to supply the main PcB with 5V !

Alexbox364, 2024-03-19:
And which Fanatec wheel base do you use? 
because my last wheel was a clubsport DD 12Nm and my PCBs weren’t working with this base while they were working fine with my DD1.
Certainly a SPI timing issue.

Lele1494, 2024-03-20:
[unlock-shell.zip](https://github.com/Alexbox364/F_Interface_AL/files/14661698/unlock-shell.zip)
Here is the 3D print files.

Alexbox364, 2024-04-15:
Hi, thank you for sharing, I’ll add the files to the repo with your credit.

closed 2024-04-15 by Alexbox364

JuanTrev24, 2024-04-25:
@Lele1494 can you help me ?