Alexbox364/F_Interface_AL #10: How to contact you

QingXY57, 2024-04-07:
Hi Alexbox364
  How do I contact you? Email？or Discord?

Alexbox364, 2024-04-15:
Hi, if it’s a question about the project you can post it here so that everyone can benefit from it.
Else discord Alexbox364, but not much active.

QingXY57, 2024-04-15:
Enable led ,OLED,TM1637,compilation will give an error for not declaring variables in advance? Or declare variables globally?
Translated by translation software

Alexbox364, 2024-04-15:
Hi, did you well install the associated libraries in your arduino IDE ahead with library manager?

QingXY57, 2024-04-15:
Associated libraries installed，After compiling, the prompt ：Compilation error: 'crc8Stat' was not declared in this scope

Alexbox364, 2024-04-17:
Hi, I'll have a look

Alexbox364, 2024-04-17:
I moved the crc8stat variable as a global variable and added pins declaration for the TM1637. Should be OK to compile. Depending on what you want to achieve with LEDs and screens, you may have to develop a bit of code.

QingXY57, 2024-04-17:
Okay, thanks.