Alexbox364/F_Interface_AL #11: NEW UNLOCKER

Lele1494, 2024-04-23:
I have re optimized the PCB and now it is a single-layer design, thinner.
All parts can be found on the DC-DC boost board and Arduino nano.
[FFB UNLOCKER V2 PCB.zip](https://github.com/Alexbox364/F_Interface_AL/files/15072151/FFB.UNLOCKER.V2.PCB.zip)
[3dPrint.zip](https://github.com/Alexbox364/F_Interface_AL/files/15072162/3dPrint.zip)

[image]
[image]
[image]
[image]
[image]
[image]

Alexbox364, 2024-04-23:
Hi! Looks awesome! Good job 👍🏻.
How much costs the assembled PCB-A from Jlpcb or equivalent?

3D_DC._2024-04-23.png (view on web)<https://github.com/Alexbox364/F_Interface_AL/assets/149867907/41183443-07e4-4697-8902-6d6f985342aa>
3D_DC.2_2024-04-23.png (view on web)<https://github.com/Alexbox364/F_Interface_AL/assets/149867907/30f837bb-7b28-454a-8d69-a51e567f2110>
_20240423140729.jpg (view on web)<https://github.com/Alexbox364/F_Interface_AL/assets/149867907/aba4bac1-bd2d-4813-a6a2-2b2e4f736bc5>
_20240423140733.jpg (view on web)<https://github.com/Alexbox364/F_Interface_AL/assets/149867907/68d3f533-086d-43cc-901e-180859c2a2b6>
_20240423140737.jpg (view on web)<https://github.com/Alexbox364/F_Interface_AL/assets/149867907/06c1e2a4-7ec8-4271-bf3e-78b27eaf4981>
_20240423140740.jpg (view on web)<https://github.com/Alexbox364/F_Interface_AL/assets/149867907/d14cbfa2-7d9a-4c21-99bd-007b3f76f2b9>

Lele1494, 2024-04-23:
I live in China, there is a PCB manufacturer called Jialichuang. PCB customization is free of charge, and if you spend money on customization, it does not exceed 20RMB. Including other parts, the cost does not exceed 50RMB

Lele1494, 2024-04-23:
I live in China, there is a PCB manufacturer called JLCPCB. PCB customization is free of charge,
 and if you spend money on customization, it does not exceed 20RMB. Including other parts, the cost does not exceed 50RMB

johnysed, 2024-04-29:
Hey, do you have the schematics, or at least components list?
What ATMEGA is it exactly? Thanks

Lele1494, 2024-04-30:
> 嘿，你有原理图，或者至少是元件列表吗？它到底是什么ATMEGA？谢谢

[image]
[image]
they all can find on the arduino nano and dc-dc booster. Note that the chip must be mega328p,not the 328pb

Krakane, 2024-05-03:
hello nice job I would like to do it I have a few questions and a problem. to start the pcb file don't go to jlcpcb when I add the compressed file it remains blocked at 96% I don't know why with different browser it's the same. If I understood correctly, it is necessary to unsolder all the components of the two cards? and where to find the connection pins you use. ? thank you in advance for your feedback

Lele1494, 2024-05-04:
The PCB is labeled with the original model number, so you only need to remove the parts you need. You can program the chip and then remove it, or you can install it and program it again. The ICSP programming interface is reserved.
The connection pins like in the picture
[image]

Krakane, 2024-05-04:
thank you for the answer on the other hand I apologize but to unlock the FFb you have to program the chip? (that was my question, I don't know if I was clear at the beginning) and if that's the case where can I find the code?

johnysed, 2024-05-06:
> thank you for the answer on the other hand I apologize but to unlock the FFb you have to program the chip? (that was my question, I don't know if I was clear at the beginning) and if that's the case where can I find the code?

The code is in the parent folder in this github.

Krakane, 2024-05-06:
> > thank you for the answer on the other hand I apologize but to unlock the FFb you have to program the chip? (that was my question, I don't know if I was clear at the beginning) and if that's the case where can I find the code?
> 
> The code is in the parent folder in this github.

ok i understand i try that when i recevied my composant thank you

xFoxHound, 2024-05-10:
bonjour , je ne parviens pas a trouver le bon fichier zip avec le pcb en monocouche , pourriez vous me donner les informations  nécessaire ? Merci beaucoup

Krakane, 2024-05-20:
[image]
hi I soldered my pcb beforehand I had programmed my chip on my arduino card and I unsoldered it to put it on the pcb as soon as I plug in my dd2 base the screen goes out and "no" steering wheel detected

johnysed, 2024-05-25:
Yeah, also can't get it to work. The comm LED is lit up, program is running but still no detection of the wheel.

Lele1494, 2024-05-28:
> ![IMG_1583](https://private-user-images.githubusercontent.com/34950473/332118380-a5bb8fb8-43d2-414a-bfa7-7038861ed397.JPEG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5MTI3NDQsIm5iZiI6MTcxNjkxMjQ0NCwicGF0aCI6Ii8zNDk1MDQ3My8zMzIxMTgzODAtYTViYjhmYjgtNDNkMi00MTRhLWJmYTctNzAzODg2MWVkMzk3LkpQRUc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyOFQxNjA3MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YjYyNjk4YzJiNTQ0Y2FhN2MxMzFjNWNlN2Q3ZGE0ZDcwMDc1MDhlNWQ5Njc0ZWU3OTdmYzRhNzJhMjhlNTgyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JB-RSqdfKxaYlGnQZZl2FiHjTQNUA4lYYj4Co4lZfKY)嗨，我事先焊接了我的 pcb，我已经在我的 arduino 卡上编程了我的芯片，然后我拆下它以将其放在 pcb 上，一旦我插入我的 dd2 底座，屏幕就会熄灭并检测到“无”方向盘

must flash the bootloader first,then flash th code.

Lele1494, 2024-05-28:
> Yeah, also can't get it to work. The comm LED is lit up, program is running but still no detection of the wheel.

must flash the bootloader first,then flash the code.

Alexbox364, 2024-05-28:
Hi, to have the at mega to communicate with the base quickly enough, you have to flash the code with ICSP port.
When I had the base going out it mainly was because the at mega wasn’t firing the spi communication quick enough at startup

johnysed, 2024-05-29:
> > Yeah, also can't get it to work. The comm LED is lit up, program is running but still no detection of the wheel.
> 
> must flash the bootloader first,then flash the code.

Why would I have to flash the bootloader first? Im trying to run the code without bootloader, so it's as fast after start as possible.
I used ICSP programmer.

Alexbox364, 2024-05-29:
Do you still have a serial output?
And what is you wheelbase?

Lele1494, 2024-05-30:
> Why would I have to flash the bootloader first? Im trying to run the code without bootloader, so it's as fast after start as possible. I used ICSP programmer.

At first, I did the same thing, but it didn't work. Later, I found out that the blank chip had to be written into the bootloader and then uploaded through the programmer to work.I'm not sure about the specific reason, maybe it's because we need raw partitions or something。

Krakane, 2024-05-30:
Thank you for your information, do you have a tutorial to follow to do this? I know a little about arduino but I admit I don't understand everything ;p

johnysed, 2024-05-30:
> At first, I did the same thing, but it didn't work. Later, I found out that the blank chip had to be written into the bootloader and then uploaded through the programmer to work.I'm not sure about [...]

Burned the bootloader and loaded in the program. Now it's not replying fast enough. Are you running it with the bootloader in?

johnysed, 2024-06-02:
> At first, I did the same thing, but it didn't work. Later, I found out that the blank chip had to be written into the bootloader and then uploaded through the programmer to work.I'm not sure about [...]

I loaded the bootloader in, but now I have stability issues. It successfully connects like 1 out of 10 times. 
How is your connection stability?

Lele1494, 2024-06-07:
> I loaded the bootloader in, but now I have stability issues. It successfully connects like 1 out of 10 times. How is your connection stability?
You try to grind this needle off by 1mm according to the picture
[image]

RoGPoWeR, 2024-07-06:
Hi Lele, idk if you are italian or not but i have a question for you, how i can program the chip if i buy it as spare? If i build all the pcb is there a method to program through the pins? Thanks

Lele1494, 2024-07-10:
> Hi Lele, idk if you are italian or not but i have a question for you, how i can program the chip if i buy it as spare? If i build all the pcb is there a method to program through the pins? Thanks

The 6pin port on the pcb is ISCP program port.

RoGPoWeR, 2024-07-10:
> Thanks

And wich firmware do i have to use? Is the stm32 usb programmer good for programming it?

RoGPoWeR, 2024-07-13:
> > Hi Lele, idk if you are italian or not but i have a question for you, how i can program the chip if i buy it as spare? If i build all the pcb is there a method to program through the pins? Thanks
> 
> 
> 
> The 6pin port on the pcb is ISCP program port.

And wich firmware do i have to use? Is the stm32 usb programmer good for programming it?

TFYR-RTY, 2024-11-14:
>我住在中国，有一家名为JLCPCB的PCB制造商。PCB定制是免费的，如果你在定制上花钱，它不会超过20元人民币。包括其他部分在内，成本不超过50元人民币>我住在中国，有一家名为JLCPCB的PCB制造商。PCB定制是免费的，如果你在定制上花钱，它不会超过20元人民币。包括其他部分在内，成本不超过50元人民币老板有qq联系方式吗

jz5588, 2024-12-11:
If I want to save the USB cable of my wheel, I can connect the USB output of my wheel to pin 9 (USB charge 5V) pin 11 (USB charge GND) pin 10 (DataPort1) pin 12 (DataPort2) through the FANANTEC base to the PC, because the pins look like USB 5V GND DP DN

jz5588, 2024-12-11:
[image]

creagel, 2025-05-15:
Is it possible to get BOM and CPL file?