Alexbox364/F_Interface_AL #12: BOM

maximilianschulte, 2024-04-30:
Could you make the BOM's (especially for ATMEGA328P_NO_HEADER) more specific. There are some parts that are not described, espacially the crystal oscillator and the capacitors. I'd also be interested in the 12 pin connector used as the fanatec connector. I'm quiet new to all that stuff, so sorry in advance if i missed anything.

Alexbox364, 2024-04-30:
Hi, to help you start, AliExpress kits for ATMEGA328p are a good base with included oscillator

https://a.aliexpress.com/_EJlHjkn

About the Fanatec connector you mean the QR pins of the 12x 2mm pitch pins on the PCB?

maximilianschulte, 2024-04-30:
No i mean the white connector comming from the ribbon cable. Thanks for the Atmega Kit, already ordered it.

Krakane, 2024-05-20:
Hi , i don't find the 4n7001 diod for full pro micro have you a link ?

Alexbox364, 2024-05-29:
Hi, sorry for late reply, there’s a mistake in item code, it’s 1N4001 diod

Alexbox364, 2024-05-29:
Its role is to have a voltage drop of the 5V to 3,7v on miso line

closed 2024-06-06 by Alexbox364