Alexbox364/F_Interface_AL #13: Help

JuanTrev24, 2024-05-02:
Hello 
I made ATMEGA328P_WITH_HEADER hoping to make it work but the programming is quite difficult for me it has caused me a big headache and I would like to use my wheel but I find it impossible is there any way even just to unlock the ffb ? Is there any way to just insert the code or something like that ? 
Thanks

JuanTrev24, 2024-05-02:
[image]

Alexbox364, 2024-05-02:
Hi, for this version of the FFB unlocked, you will need to flash the code in the ATMEGA via ICSP method.

You need another arduino uno / nano / … to do this. I placed a link in the description to help you do it.

JuanTrev24, 2024-05-03:
Thank you for taking the time to respond to me. This is the problem I'm facing for now.[image]

JuanTrev24, 2024-05-07:
I don't know what happened, I just know that it works.[image]

Alexbox364, 2024-05-07:
Hi! Good news. Problem was on uploading the code to the at mega?

Krakane, 2024-05-20:
Hi , i don't find the 4n7001 diod for full pro micro have you a link ?

closed 2024-05-29 by Alexbox364