Alexbox364/F_Interface_AL #15: Base disconnecting from PC

AVM88, 2024-07-07:
Used ful NANO,programed it with full nano in wheel_cfg. 
but when i insert it in bese its just disconnect from PC.
double checked connector wiring...

RoGPoWeR, 2024-07-07:
Have you used dc-dc boost?

AVM88, 2024-07-07:
If u mean 3v to 5v - yes.

AVM88, 2024-07-07:
Voltage is ok,in 2.98v out 5,1

RoGPoWeR, 2024-07-07:
Post a picture here and the pin side

AVM88, 2024-07-07:
> Post a picture here and the pin side

[image]

RoGPoWeR, 2024-07-07:
Sorry, wheel side pin…are you using the original fanatec connector?

AVM88, 2024-07-07:
> Sorry, wheel side pin…are you using the original fanatec connector?

No,printed one. Checked pin positions few times.

RoGPoWeR, 2024-07-07:
Give me a private contact if you want i can explain better

RoGPoWeR, 2024-07-07:
Anyway you have to drill one pin like this 1mm 
picture [image]
[image]

If the problem persist double check if there is a short circuit

AVM88, 2024-07-07:
Already checked for shorts,both sides of connector not shorted,when pluged without arduino

AVM88, 2024-07-07:
[image]

AVM88, 2024-07-07:
Even made MICRO_NO_HEADER,but same problem.

RoGPoWeR, 2024-07-07:
You dont have to remove but if you see my photo one pin is short than others [image]

From right side the second one

RoGPoWeR, 2024-07-07:
Sorry, how you are programming the chip?

AVM88, 2024-07-07:
> Sorry, how you are programming the chip?

programed with arduino IDE

RoGPoWeR, 2024-07-07:
I'm seeing that red wire connected to ground, it's the ground wire of the connector?

AVM88, 2024-07-07:
[image]
checked it  few times,visual and with multimeter

RoGPoWeR, 2024-07-07:
Have you selected the right board in arduino ide in wheel.h section?

AVM88, 2024-07-07:
[image]

AVM88, 2024-07-07:
Polarity of diode checked.

RoGPoWeR, 2024-07-07:
If disconnect when you plug the arduino i think there is a problem with a circuit, the code is correct if you have selected full nano, is your board full nano?

AVM88, 2024-07-07:
> If disconnect when you plug the arduino i think there is a problem with a circuit, the code is correct if you have selected full nano, is your board full nano?

Yes i was trying to build MICRO_WITH_HEADER,but ahd same problem.

RoGPoWeR, 2024-07-07:
Micro is different than nano, you can only flash nano board

AVM88, 2024-07-07:
[image]

Even tryed this one,and it was same... then i checked connector again,everithing seems ok.

AVM88, 2024-07-07:
> Micro is different than nano, you can only flash nano board

Micro with micro code and micro wiring.

RoGPoWeR, 2024-07-07:
Yes but the chip on the board is different

AVM88, 2024-07-07:
> Yes but the chip on the board is different

I know,but there is option supporting it....

Alexbox364, 2024-07-07:
Hi! I had this issue when the arduino did not boot quick enough at startup.
Is it working if you supply 5V from usb cable?

Alexbox364, 2024-07-07:
And which wheelbase do you use?

AVM88, 2024-07-07:
> And which wheelbase do you use?

CSL DD

RoGPoWeR, 2024-07-07:
Hi Alex, do you know what sketch i have to upload on the board of lele? And can i upload without usb?

AVM88, 2024-07-07:
> Hi! I had this issue when the arduino did not boot quick enough at startup. Is it working if you supply 5V from usb cable?

no working with USB,do i need to desolder power from step up voltage,and try then?

RoGPoWeR, 2024-07-07:
Have you desoldered the two resistors from dc-dc boost? It came in 12v

AVM88, 2024-07-07:
> Have you desoldered the two resistors from dc-dc boost? It came in 12v

Yes

AVM88, 2024-07-07:
[image]

AVM88, 2024-07-07:
cathode to wheel side

RoGPoWeR, 2024-07-07:
Is the right diode?

AVM88, 2024-07-07:
1N4001

RoGPoWeR, 2024-07-07:
Can i see all the connection/soldering between all the boards?

AVM88, 2024-07-11:
yellow wire mesure 3v,but wiring diagram

> ![Untitled](https://private-user-images.githubusercontent.com/18098097/346362710-b0ee9905-2933-4c27-a663-0e035ef16b5c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA3NDEzMDMsIm5iZiI6MTcyMDc0MTAwMywicGF0aCI6Ii8xODA5ODA5Ny8zNDYzNjI3MTAtYjBlZTk5MDUtMjkzMy00YzI3LWE2NjMtMGUwMzVlZjE2YjVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzExVDIzMzY0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0MDc4ZmNjYzExZGQ5ZmM3YmZlZmNlMmE0YjBmNWZlOTUyODcwZWY1Y2IzOWRiY2IyMTFjMTM4Nzk4Yjc0YzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tNJn-FbpZMpqhLoHGnLODXHC2he2N6EwiFMuQ6MJByk) checked it few times,visual and with multimeter

Yellow wire show 3v on multimeter,but on diagram it is  gnd together with orange wire (pin 3 and 5).  what i am missing?

AVM88, 2024-07-11:
something wrong with me or this wiring picture. :)

AVM88, 2024-07-11:
when i unsolder yellow wire,no more base disconnecting from PC (short was between orange and yellow)

RoGPoWeR, 2024-07-17:
I have your same problem but with MICRO_WITH_HEADER, when i plug into csl dd the base disconnecting from pc, have you solved the problem? I've tryed to burn bootloader, upload code, upload code with isp but no result

RoGPoWeR, 2024-07-18:
> when i unsolder yellow wire,no more base disconnecting from PC (short was between orange and yellow)

Just unsoldered that wire and no disconnection but no wheel recognised

RoGPoWeR, 2024-07-18:
I have an update, builded a nano without header and if i plug usb in the nano it's recognised by the base as a podium but without usb connected when i power up the base it disconnect

RoGPoWeR, 2024-07-20:
No one can help?

RoGPoWeR, 2024-08-09:
> when i unsolder yellow wire,no more base disconnecting from PC (short was between orange and yellow)

It's not a short issue but after milion try i've figured out the problem is the nano, i've buyed many type of nano but only one type is good for me and work perfectly on all the pcb's!

AVM88, 2024-08-09:
Which one is it,what is difference?

RoGPoWeR, 2024-08-09:
Idk what is the difference but the chip is atmega328p and not PB and you have to flash the cod through icsp port and not by usb

RoGPoWeR, 2024-08-09:
> Which one is it,what is difference?

Add me on discord my nick is rogpower we can help each other, i have found the brand of the board

For ALL, all type of code is detected as PODIUM HUB (FULL_NANO also) and the podium hub haven't the 7seg display, leds and other stuffs, tryed this stuff on full nano without chance to get it work

rcalleso, 2024-12-30:
> Add me on discord my nick is rogpower we can help each other, i have found the brand of the board
>
> For ALL, all type of code is detected as PODIUM HUB (FULL_NANO also) and the podium hub haven't the [...]

Hi RoGPoWeR, still dealing with this project? I've added you on discord

J3161, 2025-01-16:
Hi everyone can someone explain to me what code I need to run to make the atmega328pu with header work  plz. Thanks