Alexbox364/F_Interface_AL #18: Cubsport DD+ compatibility

dexter323i, 2024-12-22:
Hi!

Thank you for your awesome work! I’ve made a few wheels using this project with slight modifications for friends, and everyone is thrilled with the results! :) I decided to emulate the Porsche 918 RSR wheel because it has the most buttons.

One of my friends sold his wheel to someone with a new Fanatec DD+ wheelbase, but it didn’t work! When the wheelbase starts, it begins the initialization process but stops after 1-2 seconds. That’s it. The Fanatec app on Windows shows no wheel is attached. However, the wheel still works perfectly on other wheelbases.

Has anyone else experienced issues with the DD+? I don’t have access to this wheelbase, so I’m not sure what to check.

RoGPoWeR, 2024-12-22:
Hi, have you uploaded sketch as 918 rsr?

dexter323i, 2024-12-22:
I am not sure I understand your question. In my outgoing Fanatec messages the header is 0xA5, and the id is 0x03

RoGPoWeR, 2024-12-22:
Try 0x0C (is the podium hub) and check in fanatec interface tab in the arduino ide if the hex correspond to podium hub, i can't remember now and let me know if it work

dexter323i, 2024-12-24:
As I said, I don't have a DD+ base. One of my friends ordered it, but it will arrive only in January. :(
That is why I was asking for experiences from others here, maybe...

Bob2345de, 2024-12-31:
Had a similar issue using code designed for the older belt drive fanatec bases on a dd1. Turned out that the newer bases expected spi communication to start-up faster than the arduino could boot. Options were to use faster arduino/Mcu or remove arduino bootloader which speeds up boot process. DD+ could be expecting even faster spi startup. See here: https://github.com/lshachar/Arduino_Fanatec_Wheel/issues/14#issuecomment-1998003246

Alexbox364, 2025-01-08:
Hi everyone!
I made the move to jump from a DD1 to a DD+ Fanatec base and also had my wheels unable to work. It tried to find the cause of this misfunction by tweaking / sneaking the SPI protocol but could find the solution. 
for this reason, I switched to Simagic and had all my wheels converted to Bluetooth and works like a charm.

dexter323i, 2025-03-10:
Some update: I converted the code to Attiny414, and left only the parts I needed for my project. Still didn't work. Then I hooked up a logic analyzer, and found out the problem!
The base only sends two 33 byte messages to the wheel. If the second reply is still not correct, then it is "game over". This is the smaller issue, the larger one is the SPI interrupt routine takes too long! I put a debug I/O pin flip at the end of the routine, and it only flipped the pin, when the base was already sending the second byte! So the solution was to make the routine as quick as possible!
I cannot check it with a Nano. If you can, then try this:

```
/////////////////////////////////////////////
// SPI interrupt routine
ISR(SPI_STC_vect)
{ 
  #ifdef AVR_BOARD
    data_in.raw[isrIndex] = SPDR;
    SPDR = data_out.raw[++isrIndex];
  #endif
}
```

graydonhall, 2025-06-25:
Alexbox, if you are still monitoring this repo could you please answer this question. Does the latest code in the repo work with the Nano ESP32 (at least for your old wheelbase)? Looking at the pins.h file under the ESP32 section most of the pin definitions are just set to "1". There is also a lot of code that only runs if AVR_BOARD has been defined and the ESP32 is not AVR. I'm not a programmer and I'm trying to absorb and make sense of as much of this as I can but I will only have a chance of success if I know this code has worked at some point in time.

graydonhall, 2025-09-07:
If your nano still has the bootloader installed, there is a delay between the board receiving power and the sketch executing. This delay is too long for the wheelbase. You need to upload the sketch with a programmer instead of just using Arduino IDE to upload the sketch. Have a look here:

https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP/

What wheelbase are you using?

J3161, 2025-09-07:
Hi I'm using a csl elite as a donor base to get everything working but my main wheel is the DD pro. Here are the shunts I soldered in marking red 

[image]

J3161, 2025-09-07:
Graydonhall are you in discord I'm trying to send a video here but it won't upload. Thanks

graydonhall, 2025-09-07:
Yup, Graydonculous